Phillip-W/entwurf_vhdl_prozessor #3: generic Parameter

flomaurer, 2015-05-15:
Unsere Parameter lagern wir in ein Package aus?
Darauf müssen wir ja öfter zugreifen und damit ist der Prozessor flexibler gestalltbar.
(Also meine Meinung)

Phillip-W, 2015-05-15:
Stimme Zu
Die kann dann auch gleich jemand anlegen

flomaurer, 2015-05-15:
Ich hab vor dem Namen meiner Entity en_..._... geschrieben.
Sprich wer die architecture dazu macht sollte ar_..._... oder so machen.
Und beim package halt auch im dateinamen pack_system_... oder sowas, damit man immer gleich weiß, um was für einen typ es sich handelt.

Sabbuth, 2015-05-15:
ja immer alles in das selbe reinschreiben und aktualisieren. Das ist sehr wichtig

flomaurer, 2015-05-16:
Hi Patrick,
die 2 Dateien, die du angelegt hast gibts schon.
Es gibt außer dem Masters zweig noch einen für die CPU. und da sind schon an Haufen dateien angelegt. 
Da ist auch von der Readme eine Kopie, in der schon paar namen für die Variablen eingetragen sind, damit das einheitlich wird.
Schau dir das evtl mal an.
Ich formatiere mal über deinen Text drüber.

closed 2015-05-18 by flomaurer