Phillip-W/entwurf_vhdl_prozessor #20: IO Patrick

flomaurer, 2015-05-22:
Uns fehlt noch das komplette IO aus der letzten Vorlesung;
das müssen wir gemeinsam machen -schaff ich heut nicht mehr.

flomaurer, 2015-05-24:
Conversionpack fürs IO der OPCodes (cmd_image)

Sabbuth, 2015-05-31:
ok, ich weiß nicht, wer was zu diesem Thema tun soll... wir sollten es diskutieren

flomaurer, 2015-05-31:
Hi,
eigentlich hab ich da letzte Woche einfach ein jedem eine Funktion gegeben.
Jetzt hab ich gesehen, dass irgendwer so gut war und Integer'image gefunden hat.
Schau mal in den Masterbranch, da hab ich das auch nochmal kommentiert, weils nicht überall so geht.
Jetzt ists eh schon weniger.
Bereden wir heut um 6 - wenns früher infos brauchst ruf mich einfach auf Skype an - bin online

closed 2015-06-07 by flomaurer