Phillip-W/entwurf_vhdl_prozessor #30: Skype Session

Phillip-W, 2015-05-27:
Hey Zusammen,
also ich würde gern heut abend nochmal eine Skype Session ansetzen.
Habt ihr alle um 21:00 Zeit?

flomaurer, 2015-05-27:
geht klar

Sabbuth, 2015-05-27:
Servus... Ich bin heute um 6 in München angekommen... Danach war ich in der Arbeit bis um 5... Ich bin todmüde und habe noch gar nichts gemacht... Ich nehme gerne Teil, wenn ihr etwas zu melden habt, aber selber hab ich eigentlich nichts zu sagen. Ich schreibe die Opcodes heute und morgen...

Phillip-W, 2015-05-27:
Also bei mir war es das selbe nur bin ich gestern angekommen und heute arbeiten...
ich wollte eigentlich nur wissen ob sich irgendwas an den OP-Codes ändert.
Und ob am Freitag Entwurf is, in einer EMail hieß es irgendwas das es ausfällt?

leizhangde, 2015-05-27:
Ich habe die Zeit. Aber Skype funtioniert in meiner Wohnung gar nicht. Ich will nur wissen, was ich machen soll, weil wir extra Speicher nicht mehr brachen. Ich schreibe auch Op-Code oder was ? Es wäre besser, wenn wir eine Zeit finden und uns nochmal treffen können.

flomaurer, 2015-05-27:
Hi Leute,

also Freitag entfällt.

Wichtig wär nur, dass wir vorwärtskommen.

Ich hab sau viel Code getippt und gut Kommentiert - also solltet ihr das schaffen.

Gruß FLo

Von: leizhangde
Gesendet: ‎Mittwoch‎, ‎27‎. ‎Mai‎ ‎2015 ‎19‎:‎54
An: Phillip-W/entwurf_vhdl_prozessor
Cc: Flo Maurer

Ich habe die Zeit. Aber Skype funtioniert in meiner Wohnung gar nicht. Ich will nur wissen, was ich machen soll, weil wir extra Speicher nicht mehr brachen. Ich schreibe auch Op-Code oder was ? Es wäre besser, wenn wir eine Zeit finden und uns nochmal treffen können.

Phillip-W, 2015-05-27:
Ja dem stimme ich beim überfliegen soweit zu.

Also wichtig ist jetzt erstmal das die IO fertig wird.
Außerdem sollten bis Sonntag die OP-Codes fertig werden.
Was ja irgendwie zusammen hängt.

Soweit ich das sehen kann is der Flo mit seinen Aufgaben durch oder? 
Bitte schließt eure Issues damit jeder auch auf einen Blick sehen kann welchen teil ihr schon fertig habt.

Und da wir alle an den selben Files arbeiten werden bitte benutzt eure eigenen Branches. (Nicht das jemand unbedacht in den Master schreibt)

Also Skype Session kurzfristig abgesagt richtig? (Aus meiner Sicht hat sich eigentlich alles erübrigt)

Ich würde vorschlagen das wir dann nächste Woche mal ein Treffen an der Uni planen. Ich hab durch den neuen Aufbau irgendwie kein Gespür mehr dafür wie knapp wir drann sind...

flomaurer, 2015-05-27:
Hi,

bei mir fehlen noch die OPcodes und wir müssten in die OPCodes auch das IO reinschreiben - mach ich am Freitag.

ALso Skype ist abgesagt.

Freitag treffen bauchen wir auch nicht?
Also ich würd Skype oder so am Sonntag abend machen oder am Montag a treffen.

Gruß FLo

Von: Phillip-W
Gesendet: ‎Mittwoch‎, ‎27‎. ‎Mai‎ ‎2015 ‎20‎:‎37
An: Phillip-W/entwurf_vhdl_prozessor
Cc: Flo Maurer

Ja dem stimme ich beim überfliegen soweit zu.

Also wichtig ist jetzt erstmal das die IO fertig wird.
Außerdem sollten bis Sonntag die OP-Codes fertig werden.
Was ja irgendwie zusammen hängt.

Soweit ich das sehen kann is der Flo mit seinen Aufgaben durch oder? 
Bitte schließt eure Issues damit jeder auch auf einen Blick sehen kann welchen teil ihr schon fertig habt.

Und da wir alle an den selben Files arbeiten werden bitte benutzt eure eigenen Branches. (Nicht das jemand unbedacht in den Master schreibt)

Also Skype Session kurzfristig abgesagt richtig? (Aus meiner Sicht hat sich eigentlich alles erübrigt)

Ich würde vorschlagen das wir dann nächste Woche mal ein Treffen an der Uni planen. Ich hab durch den neuen Aufbau irgendwie kein Gespür mehr dafür wie knapp wir drann sind...

flomaurer, 2015-05-31:
Wann hättet ihr heute Zeit um uns mal zusammenzurufen?

Sabbuth, 2015-05-31:
ja heute geht es jederzeit

Sabbuth, 2015-05-31:
sagen wir mal 6?

flomaurer, 2015-05-31:
von mir aus top - ich bin die ganze Zeit verfügbar

Phillip-W, 2015-05-31:
Ich wäre für später.
Muss jetzt dann ca 2h heim fahren und des wird knapp...

flomaurer, 2015-05-31:
wann würds dir passen?
20:00 Uhr?

leizhangde, 2015-05-31:
Heute ist Geburtstag von meinem Freund. Ich muss zu ihm gehen. Mein Skype funktioniert auch nicht in dem Wohnheim.  Wenn irgendwas ich machen soll, sag mir einfach bitte.

Phillip-W, 2015-05-31:
ja 8 is super

closed 2015-06-05 by Phillip-W