Phillip-W/entwurf_vhdl_prozessor #32: Deklaration von konstanten in argumenten von Prozeduren und Funktionen

Sabbuth, 2015-05-28:
Hallo Leute,

ich vermute, dass Florian den ganzen Code bei sich zum laufen gebraucht hat... Ich kriege folgendes Problem:

Deklaration von konstanten in argumenten von Prozeduren und Funktionen brauchen bei mir ein ":" zwischen den Namen und den Datentypen. Jeder benutzt Modelsim und hat benutzt eine Standard VHDL Sprache oder? Ich hab es bei mir korriegiert, aber es ist ein bisschen Mühsam, dass wer es so übersetzen müssen...

Grüße
Patrick

flomaurer, 2015-05-31:
Hi,
also bei mir gabs keine Fehler beim Kompillieren - ich hab halt die Zeilen mit hex_image und so auskommentiert.
Grundsätzlich müssen da aber die Doppelpunkte hin - hast schon recht (hab ich teils vergessen, weiß aber nicht, warum es keine Fehlermeldung gab)
Ich werd jetzt alles nochmals durch ModelSim schieben.
Gruß FLo

flomaurer, 2015-05-31:
aktuell mit ModelSim kompillierbar

closed 2015-05-31 by flomaurer