Phillip-W/entwurf_vhdl_prozessor #75: Code gerade hochgeladen

leizhangde, 2015-06-06:
Ich habe gerade meine Code in Branch "Master" hochgeladen. Wenn es nicht funktioniert, sag mir einfach.

flomaurer, 2015-06-06:
Hi Lei,
also dein Code im def_packagebody_all funktioniert gar nicht.
Bitte probiere diesen selbst einmal zu compilen - gestern hat alle funktioniert.
Hier die Fehlermeldungen des Compilers:

# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(161): Subprogram "XSLL" parameter name "C" does not conform between subprogram declaration and subprogram body (declaration has "CO").
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(163): Prefix of attribute "length" must be appropriate for an array object or must denote an array subtype.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(164): (vcom-1136) Unknown identifier "conv_std_logic_vector".
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(165): String literal found where non-array type ieee.std_logic_1164.STD_LOGIC was expected.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(167): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(169): (vcom-1136) Unknown identifier "conv_integer".
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(171): No feasible entries for infix operator "=".
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(171): Type error resolving infix expression "=" as type std.STANDARD.BOOLEAN.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(172): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(174): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(177): Cannot read output "R".
# 	VHDL 2008 allows reading outputs.
# 	This facility is enabled by compiling with -2008.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(178): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(180): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(183): Cannot read output "R".
# 	VHDL 2008 allows reading outputs.
# 	This facility is enabled by compiling with -2008.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(184): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(186): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(189): Cannot read output "R".
# 	VHDL 2008 allows reading outputs.
# 	This facility is enabled by compiling with -2008.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(189): Cannot read output "R".
# 	VHDL 2008 allows reading outputs.
# 	This facility is enabled by compiling with -2008.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(199): Prefix of attribute "Length" must be appropriate for an array object or must denote an array subtype.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(200): (vcom-1136) Unknown identifier "conv_std_logic_vector".
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(204): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(206): No feasible entries for infix operator "=".
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(206): Type error resolving infix expression "=" as type std.STANDARD.BOOLEAN.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(207): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(209): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(212): Cannot read output "R".
# 	VHDL 2008 allows reading outputs.
# 	This facility is enabled by compiling with -2008.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(213): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(215): near "=": (vcom-1576) expecting == or '+' or '-' or '&'.
# 
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(218): Cannot read output "R".
# 	VHDL 2008 allows reading outputs.
[... cut ...] ef_package_all.vhd has no body.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(654): Procedure "XSRA" declared at line 115 of file D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_package_all.vhd has no body.
# ** Error: D:/Dokumente/Uni/Materialien/Semester 4/Entwurf digitaler Systeme/def_packagebody_all.vhd(654): VHDL Compiler exiting
# End time: 18:26:25 on Jun 06,2015, Elapsed time: 0:00:01
# Errors: 106, Warnings: 5

flomaurer, 2015-06-06:
ich fange zumindest mal an etwas auszubessern.

flomaurer, 2015-06-06:
also nachdem ich darauf jetzt echt keine Lust hab:
1) deine beiden neu eingebunden Packages müssen raus, weil die sonst mit den anderen kollidieren (Fehler verursachen und die anderen Packages das selbst können)
2) schau dir mal an, wie ich zwischen Dateitypen konvertiert hab - das geht problemlos (steht auch bei einer deiner Funktionen im Kommentar)
3) die Längen unsere arrays kennen wir - musst du nur in def_package_all nachschauen (für dich meistens data_width)

Sabbuth, 2015-06-07:
@leizhangde Lei versuche deine Sachen in unsere bisherige Arbeit einzubinden. Es ist ziemlich mühsam für einen, der deinen Code nicht kennt, den versuchen zu kompilieren. Lade dir was wir im Master Branch haben runter und baue deine Arbeit dort rein... Kompiliere ihn und lade ihn erst danach wieder hoch. Vergiss nicht, dass wir nur ein Package haben und ein Package body, indem alles drinnen steht. Wenn du eine andere nicht standard library verwendet hast, dann gib sie bitte mit einem Kommentar an, wenn du das Zeug hochlädst. Abgabetermin ist nächste Worche.

Grüße
Patrick

flomaurer, 2015-06-09:
Also Lei,
es gibt shift und rotate befehle im package nummeric_std: http://www.csee.umbc.edu/portal/help/VHDL/packages/numeric_std.vhd Dieses package ist schon eingebunden!!!
dafür musst du den interger zu einem bitvector und danach zurück wandeln:
bit_vector=to_unsigned(integer, data_width);
integer=to_integer(bit_vector);
Gruß FLo

--============================================================================
  -- Shift and Rotate Functions
  --============================================================================
 
     -- Id: S.1
  function shift_left  (  ARG: UNSIGNED; COUNT: NATURAL) return UNSIGNED;
     -- Result subtype: UNSIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a shift-left on an UNSIGNED vector COUNT times.
     --         The vacated positions are filled with Bit '0'.
     -- The COUNT leftmost bits are lost.
 
     -- Id: S.2
  function shift_right (  ARG: UNSIGNED; COUNT: NATURAL) return UNSIGNED;
     -- Result subtype: UNSIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a shift-right on an UNSIGNED vector COUNT times.
     --         The vacated positions are filled with Bit '0'.
     --         The COUNT rightmost bits are lost.
 
     -- Id: S.3
  function shift_left  (  ARG: SIGNED; COUNT: NATURAL) return SIGNED;
     -- Result subtype: SIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a shift-left on a SIGNED vector COUNT times.
     --         All bits of ARG, except ARG'LEFT, are shifted left COUNT times.
     --         The vacated positions are filled with Bit '0'. 
     --         The COUNT leftmost bits, except ARG'LEFT, are lost.
 
     -- Id: S.4
  function shift_right (  ARG: SIGNED; COUNT: NATURAL) return SIGNED;
     -- Result subtype: SIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a shift-right on a SIGNED vector COUNT times.
     --         The vacated positions are filled with the leftmost bit,ARG'LEFT.
     --         The COUNT rightmost bits are lost.
 
  --============================================================================
 
  --============================================================================
 
     -- Id: S.5
  function rotate_left  (  ARG: UNSIGNED; COUNT: NATURAL) return UNSIGNED;
     -- Result subtype: UNSIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a rotate_left of an UNSIGNED vector COUNT times.

     -- Id: S.6
  function rotate_right (  ARG: UNSIGNED; COUNT: NATURAL) return UNSIGNED;
     -- Result subtype: UNSIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a rotate_right of an UNSIGNED vector COUNT times.
 
     -- Id: S.7
  function rotate_left  (  ARG: SIGNED; COUNT: NATURAL) return SIGNED;
     -- Result subtype: SIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a logical rotate-left of a SIGNED 
     --         vector COUNT times.
 
     -- Id: S.8
  function rotate_right (  ARG: SIGNED; COUNT: NATURAL) return SIGNED;
     -- Result subtype: SIGNED (ARG'LENGTH-1 downto 0)
     -- Result: Performs a logical rotate-right of a SIGNED 
     --         vector COUNT times.

leizhangde, 2015-06-09:
Ich habe alle Code in mein Laptop kopiert. Ich habe auch meine Sache raunehmen. Aber gibt es immer 39 Errors bei def_packagebody_all.vhd. Ich kann meine Code nicht kompilieren

leizhangde, 2015-06-09:
Ich habe alle Funktion in def_Packagebody auf meinem Laptop rausgenommen. Ich verbesserte meine Code und schreibte erneut ein. Die ist kompilierbar und sieht gut aus. Deswegen habe ich gerade wieder hochgeladt. Es soll problemlos sein.

closed 2015-06-09 by flomaurer