Phillip-W/entwurf_vhdl_prozessor #76: Assembler

flomaurer, 2015-06-07:
Hi Leite, 
also ich schreib jetzt mal das Assembler Zeugs und auch dann den init Memory von Lei, weil Lei immer ohne zu testen alle hochlädt und dann nichts mehr kompilliert.
Was wir mit seinen OP-Codes machen weiß ich nicht - vielleicht spricht von euch jemand mit OHm darüber.
Gruß Flo

Phillip-W, 2015-06-07:
Wie wärs wenn du jetzt erstmal des "neue Init Memory" machst und schonmal die unterfunktionen Read_Param und Read_Register und InputDecode in den packages definierst. 

Ich würd die den Rest dann so zwischen 3 und 5 schreiben, hatte ehh vor das ich das heute mache aber ich wollte warten ob leis code noch funktionsfähig wird oder nicht...

die op codes würde ich jetzt erstmal ignorieren, des tut ja erstmal nix zur sache wenn 6 von 36 nicht funktionieren oder? darum können wir uns kümmern wenn wir noch zeit haben.

flomaurer, 2015-06-07:
Richtig, genau das mach ich grad.
aber irgendwie hab ich grad an murks drin, als ich mit der read_registers begonnen hab.
Find ich schon.

Phillip-W, 2015-06-07:
Ich meint das du des read_registers, read_param und inputDecode mir überlässt^^
sonst hast du wirklich alles gemacht und ich hab so des gefühl das ich noch wesentlich weniger gemacht hab als du und patrick

flomaurer, 2015-06-07:
k.
Dann darfst du auf fehlersuche gehen.
Viel Spaß dabei.
Vorher ists gelaufn wies geschrieben war.
Das neue Init memory geht.
Der Fehler ist in InputDecode - das read_registers ist auskommentiert

flomaurer, 2015-06-07:
Ich würd halt heut noch gern meine OPCodes testen

Phillip-W, 2015-06-07:
Ich setz mich drann.

flomaurer, 2015-06-07:
ich muss noch hochladen - art mal 15 min

flomaurer, 2015-06-07:
Ach ich lad dir das mal alles neu hoch, weil ich viel geändert hab, da auch in anderen packs und so nicht alles optimal war mit unserem IO und so.

flomaurer, 2015-06-07:
Es ist alles online.
Kannst arbeiten

closed 2015-06-07 by flomaurer