Phillip-W/entwurf_vhdl_prozessor #77: Modelsim an meinem Laptop spinnt

Phillip-W, 2015-06-07:
Ich hab grad versucht Modelsim bei mir zu installieren. (Hab bis jetzt immer an da Uni gedebugged)

Immer wenn ich ein File neu erstellen will oder öffnen will kommt folgender Error:
# wrong # args: should be "load file ?pkg? ?interp?"
hat von euch eventuell jemand erfahrung? weiß jemand was ich falsch mach?

flomaurer, 2015-06-07:
keine Ahnung, bei mir gings einwandfrei.
Du hast schon die PE Student Version von der Herstellerseite?
Dann musst noch im Browser a Formularausfüllen nach der Installation, damit du deine Lizenz bekommst und dann sollte es laufen

Ansonsten "Have you tried turning off and on again"? Sprich deinstallieren und wieder installieren

Phillip-W, 2015-06-07:
Ok nach 3 neu installationen hat es endlich geklappt :)
Dann schau ich den code jetzt schnellst möglich hin krieg.

closed 2015-06-07 by Phillip-W