Phillip-W/entwurf_vhdl_prozessor #78: Assambler-Code

Phillip-W, 2015-06-07:
Also ich hab jetzt alles gecoded.

Aber iwie schmeißt mein compiler nur fehler raus die garnicht sein können... ich glaub ich muss iwo noch packages runter laden die ich noch nicht aufm pc habe... oder ich hab iwas nicht eingebunden an das ich nicht gedacht habe... ich habe leider keine ahnung mehr was noch los sein könnte

ich kann jetzt einfach den code hochladen in dem glauben das alles stimmt oder soll ich weiter suchen?

Phillip-W, 2015-06-07:
Ich bin wirklich mit meinem Latein am ende...
ich lad den scheiß jetzt einfach mal in einem neuen Assambler#2 branch hoch und wer möchte kann da gern drüber schaun.

flomaurer, 2015-06-07:
werd ich dann heut abend mal machen - hab grad zu tun

flomaurer, 2015-06-07:
also im mem_packagebody war eine Zeile aus versehen nicht auskommentiert von dir

flomaurer, 2015-06-07:
Ich habs - auch das mit den Parametern funktioniert jetzt richtig.
Jetzt sind nur mehr alle OPCodes dazu einzufügen.
Und Leis OPCodes irgendwie zu übernehmen.

Dann heißts noch testen, testen, testen!!!

closed 2015-06-07 by flomaurer