Phillip-W/entwurf_vhdl_prozessor #81: Assembler vervollständigen

flomaurer, 2015-06-08:
Hi Patrick,
wann schreibst du den Rest meines Assembler Codes?
Ich wollt jetzt eigentlich testen.
Ansonsten schreib ich den Teil, den ich brauch selbst.
Gruß FLo

flomaurer, 2015-06-08:
Hi, ich schreib meinen Teil jetzt selber.
Schönen Abend.

Sabbuth, 2015-06-08:
Hi Flo,

ich wollte jetzt anfangen. Bist Du schon dran?

Sabbuth, 2015-06-08:
Wenn ja, lade es hoch und ich mache dort weiter.

flomaurer, 2015-06-08:
bin schon voll dabei und eig fertig - muss noch kurz 4 Fehler rausmachen.
Das war mit CopyPaste so schnell - schreibs einfach auf dich den Teil. Ist ja wurscht.

flomaurer, 2015-06-08:
schaus dir an - ist online - weiß nicht, obs richtig ist.

Ich bin die nächsten Tage nicht da. Sprich ich werde meinen Teil Testen und macht ihr bitte aus, wie das mit Leis Teil wird.

Bei meinem Teil bin ich mir mit den Flags nicht ganz sicher.

closed 2015-06-08 by flomaurer