Phillip-W/entwurf_vhdl_prozessor #84: Code abgeben

flomaurer, 2015-06-09:


flomaurer, 2015-06-09:
Hi Leute,
also wer gibt den Code am Freitag an Hernn Ecker ab?
Und wie machen wir ihm klar, wer was geschrieben haben - verwenden wir hierfür einfach ein Worddokument, in welchem steht, wer welche Funktionalitäten / Funktionen / Proceduren geschrieben hat? Oder Kopieren wir den Code in Word oder als PDF, wobei wir alle Zeilen farblich hinterlgen (jede Farbe einer Person zuordnen)?

Phillip-W, 2015-06-10:
Ich kann ihn an den Herrn Ecker weiter geben.
Ich erstelle gleich noch ein .txt file im Master in das bitte jeder schreibt was er gemacht hat.

flomaurer, 2015-06-11:
Hab mal rei, was ich weis.
schauts drüber, lest Korrektur und Testet wenn möglich nochmals ausgebig - seit Lei was geschrieben hat, hab ich mir das nicht mehr angeschaut - auch nicht probiert zu kompilieren.
Gebt eine Kompeliierbare Version ab.
Suber Danke.

Phillip-W, 2015-06-12:
Habe Email soeben verschickt.

closed 2015-06-12 by Phillip-W

flomaurer, 2015-06-12:
Top.
Ich bin heut nicht da.
Viel Spaß.
Gruß FLo