Phillip-W/entwurf_vhdl_prozessor #86: Flags setzen

Phillip-W, 2015-06-10:
Leute ich teste gerade meine Jumps und habe festgestellt das die Flags oft nicht korrekt gesetzt werden.
Ich kenne mich leider in euren Funktionen zu schlecht aus um das auf anhieb verbessern zu können.

flomaurer, 2015-06-10:
OK, habt ihr bei mir die Flags schon getestet?
Ich habs überflogen, weiß aber nicht obs immer stimmt.
Also ich hab die Flagbefehel von Hand in meinen Code geschrieben.

Ich weiß nichts von einer zentralen Flag Funktion.

das ist 3.2 in der Spec.

_________________________________________________________________________________
3.2.	Internal Flags
3.2.1.	In order to store particular results of some operations, there are 4 internal Flags: Zero, Carry, Negative and Overflow.
3.2.2.	These flags are bits that can contain the values ‘0’ (the flag is cleared) or ‘1’ (the flag is set).
3.2.3.	The flags are only affected when a data word is written into a register (that means by the following instructions: ADD, ADDC, SUB, SUBC, NOT, AND, OR, XOR, REA, REO, REX, SLL, SRL, SRA, ROL, ROLC, ROR, RORC, LDC, LDD, LDR, IN).
3.2.4.	Zero Flag
3.2.4.1.	When executing one of the instructions listed in 3.2.3, the result is evaluated; if the value equals 0x0, the zero flag is set, otherwise it is cleared.
3.2.4.2.	The initial value of the zero flag is ‘0’.
3.2.5.	Carry Flag
3.2.5.1.	The carry flag is cleared by the following instructions: NOT, AND, OR, XOR, REA, REO, REX, ROL, ROR.
3.2.5.2.	When executing the ADD or ADDC instruction, the carry output of an unsigned addition of the two source operands is assigned to the carry flag. 
3.2.5.3.	When executing the SUB or SUBC instruction, the carry output of an unsigned addition of the first source operand and the two’s complement of the second source operand is assigned to the carry flag.
3.2.5.4.	When executing the SLL or ROLC instruction, the most significant bit of the source operand is assigned to the carry flag.
3.2.5.5.	When executing the SRL, SRA or RORC instruction, the least significant bit of the source operand is assigned to the carry flag.
3.2.5.6.	LDC, LDD, LDR and IN don’t affect the carry flag.
3.2.5.7.	The initial value of the carry flag is ‘0’.
3.2.6.	Negative Flag
3.2.6.1.	When executing the ADD, ADDC, SUB or SUBC instruction, the source operands are sign-extended by one bit; the additional bit of the result is assigned to the negative flag.
3.2.6.2.	When executing one of the remaining instructions listed in 3.2.3, the most significant bit of the result is assigned to the negative flag.
3.2.6.3.	The initial value of the negative flag is ‘0’.
3.2.7.	Overflow Flag
3.2.7.1.	When executing the ADD, ADDC, SUB or SUBC instruction, the source operands are sign-extended by one bit; if the additional bit of the result differs from the (original) most significant bit of the result, the overflow flag is set, otherwise it is cleared.
3.2.7.2.	When executing the SLL instruction, the overflow flag is set if the most significant bit of the source operand differs from the most significant bit of the result, otherwise it is cleared.
3.2.7.3.	When executing one of the remaining instructions listed in 3.2.3, the overflow flag is cleared.
3.2.7.4.	The initial value of the overflow flag is ‘0’.

Phillip-W, 2015-06-10:
Naja eine zentrale Funktion brauchen wir nicht

Ich habe nur gerade mit ADD und SUB die flags verändert weil ich ja meine Jumps von flags abhängig machen muss und da hat quasi nichts gestimmt.

Außerdem sind wir glaub ich sehr sprunghaft ob wir bis max 2047 bis - 2048 oder 4096 bis 0 als wertebereich nehmen.
zum beispiel wenn ich 3000 lade mit ldc setzen wir das negative flag, wenn ich aber 2000 mit 100 addiere setzen der prozessor kein negativ flag.

Phillip-W, 2015-06-10:
Also ich habe überall die Flags eingefügt so wie sie sein müssten falls es nicht schon vorhanden war.

Patrick bitte überprüfe das nochmal, ich bin mir bei deinem Teil nicht zu 100% sicher.

Sabbuth, 2015-06-10:
Ok ich fange jetzt damit an. Ich werde sie durchtesten und korrigieren. Danach melde ich mich wieder

Sabbuth, 2015-06-10:
zum beispiel wenn ich 3000 lade mit ldc setzen wir das negative flag, wenn ich aber 2000 mit 100 addiere setzen der prozessor kein negativ flag.

Es heißt die Operanden sind sign-extended, das heißt hättest du eine 8= 1010 und eine 12= 1100 so wäre bei einer Addition beim hinzufügen eines signed bit folgende:

+ 01010
   01100

und bei einer fünfbit darstellung im signed-Bereich, hättest du folgendes Ergebnis:

   00110

mit einem Overflow= true

Deswegen soll bei deiner Addition von 2000 mit 100 keine negative Zahl entstehen!

Noch was: die Operanden, wie schon vom Anfang an definiert, funktionieren nur im Positiven bereich... Dennoch können die Lösungen(dank des Flags) negativ werden. Durch verwendung der Sub und Subc Anweisungen ist unser Prozessor trotzdem in der Lage negativ zu rechnen, obwohl keine negative Operanden direkt eingegeben werden.

leizhangde, 2015-06-10:
Ich bin gerade zurückgekommen. Ich habe meine Code geschaut, und noch bisschen verbessert. Meine Code in File en_cpu_mf ist von jemdem in führer Version zurückgesetzt. Bitte Vorsichtig. Meine Code sind zweimal von Jemendem gelöscht.
Ich werde auch nicht mehr mitmachen, weil die Semester für mich Praktikumsemester ist . Ich habe ganz knapp Zeit. Ich kann euch wiklich ganz kanpp helfen. Tut mit leid. und Viele Spaß.

Sabbuth, 2015-06-10:
Lei ich habe versucht deinen Code zu kompilieren... Es geht leider immer noch nicht. Ich musste ihn wieder auskomentieren. Hat ihn jemand schon kompilieren können? Es wäre gut wenn man es noch morgen hinkriegen würde

Grüße
Patrick

Phillip-W, 2015-06-10:
Ich habe vorher den Master geupdated mit leis code und da ging alles super.
jetzt hat der lei wieder iwas verändert jetzt gehts nimma...

mach des zeug des der lei geändert hat wieder rückgängig dann sollte alles klappen.

flomaurer, 2015-06-11:
hoffe das mit Lei bekommt ihr irgendwie auf die Reihe - mit den Befehlen die ich ihm an die Hand gegeben habt, kann man ohne zu denken (evtl. alle) Codes schreiben.

schauts drüber, lest Korrektur und Testet wenn möglich nochmals ausgebig - seit Lei was geschrieben hat, hab ich mir das nicht mehr angeschaut - auch nicht probiert zu kompilieren.
Gebt eine Kompeliierbare Version ab.
Suber Danke.

closed 2015-06-12 by Phillip-W